reuterbal/photobooth #146: Error: cannot identify image file >_io.BytedsIO object at 0x7146a570

michi1g, 2019-06-30:
**Describe the bug**

Error when starting up

> Error: cannot identify image file >_io.BytedsIO object at 0x7146a570

**To Reproduce**

Using gphoto2 app crashes when i click on "Start Photobooth"

**Expected behavior**

App starts and works like a charm ;-)

**Screenshots/Screencast**

> 2019-06-30 16:28:13,081 - root - INFO - Using camera with preview functionality
> 2019-06-30 16:28:15,455 - root - ERROR - CameraProcess: Exception "cannot identify image file <_io.BytesIO object at 0x7146a570>"
> Traceback (most recent call last):
>   File "/home/pi/photobooth/photobooth/main.py", line 66, in run
>     if cap.run():
>   File "/home/pi/photobooth/photobooth/camera/__init__.py", line 94, in run
>     self.handleState(state)
>   File "/home/pi/photobooth/photobooth/camera/__init__.py", line 101, in handleState
>     self.startup()
>   File "/home/pi/photobooth/photobooth/camera/__init__.py", line 67, in startup
>     test_picture = self._cap.getPicture()
>   File "/home/pi/photobooth/photobooth/camera/CameraGphoto2.py", line 144, in getPicture
>     return Image.open(io.BytesIO(file_data))
>   File "/usr/lib/python3/dist-packages/PIL/Image.py", line 2349, in open
>     % (filename if filename else fp))
> OSError: cannot identify image file <_io.BytesIO object at 0x7146a570>
> 2019-06-30 16:28:15,483 - root - ERROR - Camera: cannot identify image file <_io.BytesIO object at 0x7146a570>

**Hardware (please complete the following information)**

 - Device RPi 3
 - Camera Canon EOS 70D
 - GPIO: [Yes, but not enabled]

**Software (please complete the following information)**

 - OS Raspbian Stretch
 - Python version 3.5.3
 
 **Installed packages**
 
> automationhat==0.1.0
> blinker==1.3
> blinkt==0.1.2
> buttonshim==0.0.2
> Cap1xxx==0.1.3
> chardet==2.3.0
> click==6.6
> colorama==0.3.7
> cryptography==1.7.1
> drumhat==0.1.0
> envirophat==1.0.0
> ExplorerHAT==0.4.2
> Flask==0.12.1
> fourletterphat==0.1.0
> gphoto2==2.0.0
> gpiozero==1.4.1
> idna==2.2
> itsdangerous==0.24
> jedi==0.10.2
> Jinja2==2.8
> keyring==10.1
> keyrings.alt==1.3
> MarkupSafe==0.23
> mcpi==0.1.1
> microdotphat==0.2.1
> mote==0.0.4
> motephat==0.0.2
> numpy==1.12.1
> oauthlib==2.0.1
> pantilthat==0.0.7
> pgzero==1.2
> phatbeat==0.1.1
> pianohat==0.1.0
> picamera==1.13
> picraft==1.0
> piglow==1.2.4
> pigpio==1.38
> Pillow==4.0.0
> pkg-resources==0.0.0
> pyasn1==0.1.9
> pycrypto==2.6.1
> pygame==1.9.3
> pygobject==3.22.0
> pyinotify==0.9.6
> PyJWT==1.4.2
> pyOpenSSL==16.2.0
> pyserial==3.2.1
> python-apt==1.1.0b5
> pyxdg==0.25
> rainbowhat==0.1.0
> requests==2.12.4
> requests-oauthlib==0.7.0
> RPi.GPIO==0.6.5
> RTIMULib==7.2.1
> scrollphat==0.0.7
> scrollphathd==1.2.1
> SecretStorage==2.3.1
> sense-emu==1.1
> sense-hat==2.2.0
> simplejson==3.10.0
> six==1.12.0
> skywriter==0.0.7
> sn3218==1.2.7
> spidev==3.3
> thonny==2.1.16
> touchphat==0.0.1
> twython==3.4.0
> unicornhathd==0.0.4
> urllib3==1.19.1
> virtualenv==15.1.0
> Werkzeug==0.11.15

I had an older version (about a year old) running but upgraded due to new features. The old version is still working with preview etc.

Any suggestions how to fix this? I'm invited to a birthday and was asked to bring the box with me.

reuterbal, 2019-06-30:
Please delete Tour photobooth.cfg file, try again, and provide the full log.

michi1g, 2019-06-30:
Deleting config solved the issue!

closed 2019-06-30 by michi1g