reuterbal/photobooth #148: Wedding Photobox for my brother

websta911, 2019-07-02:
**Describe your photobooth**

Thank you very much for providing the code for this photobooth it really helped me a lot.
Initially I built it for my brothers wedding with the previous first version of your photobox and upgraded to the new version and added some features  for some birthdayparties and weddings in the last year. 
I made it out of plywood and engraved some birds around the camera hole (as an inside joke). Because of an older camera(no Live view) available I used the raspicam as preview camera and shot with the old canon but since then upgraded to the Canon 550D (with liveview), thats the little black hole in the front, I plan on closing that
Unfortunately I always forget to take a picture when this thing is in action so there are just shots during maintenance or upgrades.  When in use it sits on 4 legs also made from Plywood.

With every use of it I try finessing some things around that. I am planning on uploading the pictureBrowser to github, if someone likes that and wants to use it but I just made the account so please be patient.

**Hardware :**
 - Raspberry Pi 3B+
 - Canon EOS 550D
 - GPIO: No
 
 **Additional equipment**
  - 10 Inch Touchscreen
  - warmwhite and coldwhite LED Stripes with Dimmers used as illumination the possibility to set the color temperature by dimming warmwhite or coldwhite is convenient.
  - T-Mobile WlanCube
  - Sinfonia CS2 Printer
  - powered USB Hub
  - some fans
  - Z style camera mount
 - old tablet in "Kiosk" mode attached to PhotoboxWLan can browse the "pictureBrowser" shows all recent photos

**Software:**
 - Raspbian Stretch
 - Python version 3.5.3

**Modifications:**
 - Added fancy Layout options (in the Pictures) the combination of the "fancy Layout" button and the "omit last Picture" defines which layout is used. Plan is to create a seperate tab for that. 
 - add Background images or solid colorbackground with colorpicker in settings
 - add Logo to the pictures, Logo can be rotated
 - added a seperate Project with creates a website where Guests can request digital copies via Email by providing their email address.  
       - website has management functions Eg. Upload of backgrounds or Logo. 
       - initially the management functions where for the first version and I referenced the Logo and Background images by using symlinks and via the website you could change the symlink to a new logo oder background and therefore easily change them without the need for code modification.   

**Problems:**
 - Printing is a little bit laggy but I already saw the possible fix and will try to upgrade to it.

**Pictures:**
[image]
[image]
[image]
[image]

[image]
[image]

thyristor59, 2019-07-02:
Awesome box and the birds are very beautiful! 

I am amazed by the multiple choice of image layouts! would you have a plan to follow to achieve the same thing at home?

 I am too bad to do it alone. but I progress in the python thanks to this photobooth! in advance thank you so it does not bother you!

websta911, 2019-07-04:
> Awesome box and the birds are very beautiful!
>
> I am amazed by the multiple choice of image layouts! would you have a plan to follow to achieve the same thing at home?
>
> I am too bad to do it alone. [...]

OK here is the assemblePicture function from _init_.py in the cameraFolder.

I am not a programmer and I struggled alot with the math.
There is a high chance that some things should have been calculated somewhere else but as mentioned I am not very good at programming 

and for it to work you need the _Fancy setting,  background and logo.
If you want to add that to the Settings just look at Frames.py and you can reverse engineer how to put those settings in.

have fun with it 
regards

def assemblePicture(self):
        """my custom Version of assembling Pictures"""

        self.setIdle()

        picture = self._template.copy()
        if(self._fancy):

            if(self._skip_last):
                
            # -------------- 3 Rotated Pictures
                outer_border = 30
                inner_border = 30
                pic_rotation = 11

                largePics_size = ( int( (self._pic_dims.outputSize[0] - ( outer_border *4))  // 10)*4.6   ,
                                int( (self._pic_dims.outputSize[1] - ( outer_border *4 ) )//10)*4.6)
                evenlargerPics_size = ( int( (self._pic_dims.outputSize[0] - ( outer_border *4))  // 10)*5.5   ,
                                int( (self._pic_dims.outputSize[1] - ( outer_border *4 ) )//10)*5.5)
                logo_size = ( int( (self._pic_dims.outputSize[0] ) //3 ),
                            int( (self._pic_dims.outputSize[1])//2 ) )
                #logo_size = (1206, 2669) #for hardcoded use
                logging.info('size largePicx "{}"'.format(largePics_size))
                logging.info('Size logo "{}"'.format(logo_size))

                # Image 0
                img = Image.open(self._pictures[0])
                img = img.convert('RGBA')
                img.thumbnail(largePics_size)
                logging.info('Image0 inserted with size pre rotation "{}"'.format(img.size))
                img = img.rotate(pic_rotation, expand=True)
                offset = ( outer_border  ,
                        outer_border  )
                picture.paste(img, offset, img)
                logging.info('Image0 inserted with size "{}"'.format(img.size))
                logging.info('largepic size "{}"'.format(largePics_size))

                # Image 1
                img = Image.open(self._pictures[1])
                img = img.convert('RGBA')
                img.thumbnail(largePics_size)
                img_norotated = img.size
                img_norotated_small = img.size
                img = img.rotate(pic_rotation, expand=True)
                img_small = img
                ta =  (img_norotated[1]+outer_border) * math.cos(math.radians(90 - pic_rotation)) ## a = Hypothenuse c * cosinus von beta, beta = 90° - Alpha #not working weil negativ.
                tb = math.sqrt(img_norotated[1] ** 2 - ta ** 2) #b = sqrt(c² - a²)
                logging.info('tb "{}"'.format(tb))
                logging.info('ta "{}"'.format(ta))
                logging.info('ta "{}"'.format((outer_border - round(ta))))
                #offset = ( (outer_border + round(ta)),  
                        #(self._pic_dims.outputSize[1] - outer_border - img.size[1]) ) #b = sqrt(c² - a²)
                #offset = ( ((self._pic_dims.outputSize[0] - outer_border*2 ) //2) - img.size[0],
[... 135 lines cut ...]
                    logo = Image.open(self._logo).convert("RGBA")
                    logo = logo.rotate(self._logo_rot, expand=True)
                    if(self._skip_last):
                        logo_size = ((self._pic_dims.thumbnailSize[0] / 100)*90,
                                    (self._pic_dims.thumbnailSize[1] / 100)*90)
                        logo.thumbnail(logo_size, Image.ANTIALIAS)
                        offset = ((self._pic_dims.thumbnailOffset[self._pic_dims.totalNumPictures])[0]+ (self._pic_dims.thumbnailSize[0] -logo.size[0]) // 2,
                                (self._pic_dims.thumbnailOffset[self._pic_dims.totalNumPictures])[1]+ (self._pic_dims.thumbnailSize[1] -logo.size[1]) // 2)
                    else:
                        logo_size = ((self._pic_dims.thumbnailSize[0] / 100)*70,
                                    (self._pic_dims.thumbnailSize[1] / 100)*70)
                        logo.thumbnail(logo_size, Image.ANTIALIAS)
                        offset = ((picture.size[0] - logo.size[0]) // 2,
                                (picture.size[1] - logo.size[1]) // 2)

                    picture.paste(logo,offset,logo)
                    logging.info('Logo inserted')
                else:
                    logging.info('No logo  defined')

        byte_data = BytesIO()
        picture.save(byte_data, format='jpeg')
        self._comm.send(Workers.MASTER,
                        StateMachine.CameraEvent('review', byte_data))
        self._pictures = []

thyristor59, 2019-07-04:
awesome ! a big thank you
 I would do a test on my test version. 
it seems to me fun to do

reuterbal, 2019-07-05:
Thank you for sharing your story! I like the different layouts very much - I'll probably add something similar in the future.

thyristor59, 2019-07-05:
that's what everyone asked me for lotd of the wedding, different layouts, i did not have time to watch but a direct addition without modification of personal file would be perfect reuterbal

tommy1974, 2019-07-17:
Very nice
, would u share the project of the PictureBrowser?

MarioJanssen27, 2019-09-18:
> OK here is the assemblePicture function from _init_.py in the cameraFolder.
>
> I am not a programmer and I struggled alot with the math.
> There is a high chance that some things should have been [...]

In which file do I have to insert the code?
And does anything else have to be done to make it work?

--------------------------------------------------------------------
in deutsch

In Welcher Datei muss ich den Code einfügen?
Und muss sonst noch was machen damit es klappt?

websta911, 2019-09-19:
Hello, 
So I forked the latest codebase and now there should no need to second guess where the assemble code goes. 
Head over and try that. 
Upon merging my changes into the original one I tried to comment on some of the changes.

Have fun
Christoph

thyristor59, 2019-09-19:
just great!
 I just did a test, everything is perfect! 

However, what is the data to modify when I rotate the camera at 90 °?

websta911, 2019-09-19:
Good question, I never thought of that nor tried that.

But all the layout stuff for the fancy layouts is in camera/init.py in the
assemble_picture function

The offsets and sizes are guestimated there.

Best regards

MarioJanssen27, 2019-09-19:
> Good question, I never thought of that nor tried that. But all the layout stuff for the fancy layouts is in camera/init.py in the assemble_picture function The offsets and sizes are guestimated there. Best regards
> […](#)

Can someone upload the new full file?

That you only have to make copyright paste?!????

That would be very nice

MarioJanssen27, 2019-09-22:
Traceback (most recent call last):
  File "/usr/lib/python3.5/runpy.py", line 183, in _run_module_as_main
    mod_name, mod_spec, code = _get_module_details(mod_name, _Error)
  File "/usr/lib/python3.5/runpy.py", line 142, in _get_module_details
    return _get_module_details(pkg_main_name, error)
  File "/usr/lib/python3.5/runpy.py", line 109, in _get_module_details
    __import__(pkg_name)
  File "/home/pi/photobooth/photobooth/__init__.py", line 21, in <module>
    from .main import main
  File "/home/pi/photobooth/photobooth/main.py", line 33, in <module>
    from . import camera, gui
  File "/home/pi/photobooth/photobooth/camera/__init__.py", line 165
    self.setIdle()
       ^
IndentationError: expected an indented block

**I get this error message when I insert the code into __init__.py. What do I have to do?**

------------
in deustch

**Ich bekomme diese Fehlermeldung wenn ich den Code in die __init__.py einfüge. Was muss ich machen?**

websta911, 2019-09-22:
Hello,

The code has to go in the assemble_pictures function. In the init.py in the
camera subfolder.

But you can check out a working version or compare with my forked version
https://github.com/websta911/photobooth

Best regards
Christoph

MarioJanssen27, 2019-09-23:
Okay, that works, but now I don't have lightning! What do I have to change or what has changed for the flash to work again????

---------------------------
In deutsch

Okay das funktioniert aber jetzt habe ich keinen Blitz mehr! Was muss ich ändern oder was hat sich geändert damit der Blitz wieder funktioniert????

websta911, 2019-09-23:
Hello,

great o hear that that worked. I guess the camera triggers the flash when
needed? or do you have that hooked up to the raspberry pi?
From the parts you are telling me here I cannot make a good guess what
could be the problem.

Does the flash trigger if you use the camera manually?
Did you make any changes to the camera?
Is it bright enough so the camera does not need a flash?
If you try your old version does the flash trigger under the same
circumstances?

If you have questions and issues it is easier to suggest solutions if there
is some kind of context the problem appears in.

I do not use a Flash in my setup I just have led Lightning, so I have no
experience with flash and the photobooth.
best regards
Christoph

MarioJanssen27, 2019-09-23:
The flash is connected directly to the camera. The flash is always off. Why don't I know until now that was never a problem. With the new version the flash only works when the Photobooth is started. With the photos there is no flash. 

If I operate the camera Manuel the flash triggers.
I made no changes to the camera.
I made test with bright and dark room both times no flash.
With the old version (SD-Backup) the flash fires.

**Can I compare a file that has changed between my version and the new version?**

Two errors are displayed during installation. "see photos"

---------------------------------------------------
in deutsch

Der Blitz ist direkt an der Kamera angeschlossen. Der Blitz eigentlich immer aus. Wieso weis ich auch nicht bis jetzt war das nie ein Problem. Bei der neuen Version arbeitet der blitz nur beim start der Photobooth. Bei den Fotos kommt kein Blitz. 

Wenn ich die Kamera Manuel betätige löst der Blitz aus.
Ich keine Änderungen an der Kamera vorgenommen.
Ich habe Test gemacht bei hellen und dunkeln Raum beides mal kein Blitz.
Bei der alten Version (SD-Backup) löst der Blitz aus.

**Kann ich eine Datei vergleichen was sich geändert hat zwischen meiner Version und der neuen Verison?**

Beim installieren werden zwei Fehler angezeigt. "siehe Fotos"

MarioJanssen27, 2019-10-03:
The cable has a loose contact

MarioJanssen27, 2019-10-06:
The autostart (--run) does not work! What has changed? Or do you have another solution?

----------------------------
in deutsch

Der Autostart (--run) funktioniert nicht! Was hat sich geändert? Oder hast du eine andere Lösung?

websta911, 2019-10-06:
Hello,
Okay do you have error messages upon starting? What are does?
Does it not start at all?
What exactly are you calling from where?
The autostart.sh? Or in the virtual environment (usually .venv) "python -m
photobooth -- run"
Are there messages in the photobooth.log?

It should work I tried it today via autostart.sh file from the root folder
of the photobooth folder.

Regards Christoph

Tinker232, 2019-10-15:
Hiiii, 
thanks for the great modifications. The fancy layouts are pretty fantastic!!!
The next step would be a further layer in the UI (e.g. after pressing the "hit me" Button) for choosing between the different layouts. Otherwise a random feature will work fine as well and would be easier to realize I guess...
But, I think the configs / settings, which also includes the layout are imported only once by startup the photobooth, isn´t it? 
Thanks Tina :-)

websta911, 2019-10-18:
Hello,. The picture Library is public now.
You can find that in my GitHub right here: https://github.com/websta911/pictureLibrary

Have fun.

Tinker232, 2019-11-06:
Hey @websta911 
it is possibile to make only a single shot without any layouts?
LG Tina

websta911, 2019-11-07:
ello Tina,
yes it is possible, you have to make some changes, in
photobooth/photobooth/camera/PictureDimensions.py

After line 80  add the if clause and else path around the for block.

In the GitHub repo the changes are already in so you can copy them out of
there.

Have fun
Christoph

AndPDale, 2021-04-21:
@websta911 - Great upgrade and the additional layouts are a good addition. 
I see in your intro description that you initially used a camera and raspberry pi camera for the preview. Are you able to share the code to enable this please, as this is a feature that would enable my setup to function? 

Currently my nikon D90 will autofocus but not show a preview on gphoto-commandline but if i change to cffi I get preview but no autofocus. So i plan to use the picamera for the preview. Thanks for your help

websta911, 2021-05-01:
> @websta911 - Great upgrade and the additional layouts are a good addition.
> I see in your intro description that you initially used a camera and raspberry pi camera for the preview. Are you able to [...]

Hi, 
I thought about that but could not find an old version with this setting in place.  But I think that is because I only did that in the first version of the photobooth where it all was depending on pygame, since the rewrite which is the current version I think I never implement that because I already had a preview capable camera. So I am sorry but I do not have a solution for you.

best regards 
Christoph

AndPDale, 2021-05-02:
> Hi,
>
> I thought about that but could not find an old version with this setting in place.  But I think that is because I only did that in the first version of the photobooth where it all was [...]

Hi Christoph, thank you for your reply and explanation. It sounds like you managed to use a D90 with preview and autofocus. Which photo booth setting did you use to achieve this? I can get autofocus on command line option but no preview. 
Thanks
Andrew