reuterbal/photobooth #149: Portable photobooth to print 2 identical photo strips

ctrlw, 2019-07-06:
Thanks for this project! I used it to build a photobooth for a wedding, with 2 main goals:
* It should print 2 identical strips of photos, like traditional photobooths, so one could be stuck into a guest book, and the other be kept by the guest
* It had to fit into my hand luggage

It was a success, the couple got a nice guest book, people had fun and were queuing up to take pictures, and some even said it was more fun than the other photobooths they had seen at weddings. That's great, as it took a lot of time preparing this and babysitting it at the party.

**Hardware**
 - Device: Raspberry Pi 3B+
 - Camera: Raspberry Pi camera v2
 - GPIO: Yes (single button)
 
 **Additional equipment**
 * 10 W LED light (20 Euro): https://www.amazon.de/gp/product/B07D9FHG8T
 * 5" resistive touch screen (36 Euro, doesn't need to be touch): https://www.amazon.de/gp/product/B01F3JI58G/
 * Button from arcade set
 * Wooden box 23x23x8cm (originally contained 3 beer bottles) with custom front
 * Canon Selphy 1300 (112 Euro, connected with a spare USB cable): https://www.amazon.de/gp/product/B074QH7NFF
 * Cutting machine / Hebelschneider
 * Disguise material (hats, glasses, rubber chicken, ...) and backdrop: https://www.amazon.de/gp/product/B07FD58RBS
 * For debugging keyboard, mouse, USB hub
  
**Software**
 - OS: Raspbian Stretch
 - Python version: 3.5.3
 - Modified photobooth code: https://github.com/ctrlw/photobooth

**Modifications:**
 * Takes 4 pictures, assembles them into 2 identical photo strips
   * Adapted picture sizes, margins and photo resolution accordingly
   * I used the development branch, but didn't understand the margin handling, so I simplified it for my purpose
   * Set print size to 1232x1823, which seems to be Selphy's native resolution: https://www.dpreview.com/forums/thread/1913260
 * Small box, to fit into hand luggage
 * Simple usage: single button press to start, no other user interaction
 * Better scaling algorithm: with the original code, scaled photos looked pixelated. Switched to bicubic interpolation and now it looks much better
 * Reduced waiting times. A little stress makes the pictures more fun
 * Changed some GPIO pins, as the Raspberry Pi is directly stuck onto my screen via GPIO
 * Automatically boots up into photo taking mode: https://www.raspberrypi-spy.co.uk/2014/05/how-to-autostart-apps-in-rasbian-lxde-desktop/
`sudo nano /etc/xdg/lxsession/LXDE-pi/autostart`
appended following line
`/home/pi/photobooth/autostart.sh`

**Problems:**
 * Light: I should have used a flash instead, as the light is so bright that it disturbs people, but not bright enough to avoid blurred pictures
 * I had to stay close to the photobooth to change paper / ink, as the printer holds only paper for 18 photos at once
 * After changing, the printer driver would not realise that printing was ok again, so I had to manually run `cupsenable Canon_SELPHY_CP1300`
   * Having the keyboard available, allowed to print some extra photos on request `lp <filename>`
 * I fixed the screen inside with velcro, which is not very stable
 * I added no cue like shutter noise or flash when the photo is taken, so people had to look at the screen, which sometimes resulted in unwanted movement
 * Photo strips had to be cut by hand, with varying results
 * My screen resolution is only 800x480, which is not high enough to show the buttons, at least in dark mode. I resorted to editing photobooth.cfg by hand

**Pictures:**
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

reuterbal, 2019-07-08:
Thanks for sharing!

I have resolved most printing issues by switching to pycups and connecting the CP1300 via wifi using the airprint protocol. Have you tried that?

ctrlw, 2019-07-14:
No, but I'll try pycups the next time. I'm surprised that wifi is more stable here than a USB connection.

From my local modifications I'd recommend to take the bicubic image resizing into the main repository, and created a pull request. This greatly improved resulting image quality, e.g. for people with glasses. I didn't notice a performance impact, but my original photos were already very small (640x480). If performance is a concern, maybe it could be offered as a configuration.

reuterbal, 2019-07-17:
The main difference between USB and WiFi printing is the used PPD file: with WiFi, the picture file is sent to the printer and processed there for printing while with USB, the picture file is processed on the RPi and then send to the printer. I think that is responsible for some of the delay and is the reason that printing via Wifi yields a better quality.

We should definitely change to bicubic resizing, thanks for that!

thyristor59, 2019-09-26:
Hello ! 
what did you change to have 2 identical band?

 I can not do it maybe I'm too stupid .....

 I have a single strip in the middle of the sheet print! 
it may be due to the 90 ° rotation of the camera?

ctrlw, 2019-09-26:
Hi thyristor59,

Did you use my branch? I did a few changes based on the official development branch at the time. You can simply take it and it should directly work, at least if you have a similar setup: https://github.com/ctrlw/photobooth/tree/development

You can also just view my code changes and adapt the code yourself:
https://github.com/reuterbal/photobooth/compare/development...ctrlw:development

thyristor59, 2019-09-27:
I managed to have 2 identical bands! thank you !
 now the format is not excellent! 
Remember the settings of the application you had changed? 
in advance again thank you!

ctrlw, 2019-09-27:
The settings are also in my branch, as I changed the default settings file.

I used the selphy’s standard format 10x14.8cm prints, and  640x480 images
from the raspicam.

thyristor59, 2019-09-27:
I work it ! All it's good for me !
Thx!!