reuterbal/photobooth #150: autofocus -

knutta, 2019-07-08:
I've installed the photobooth on my RaspberryPI 3. It's a clean install, and is just a straight up step-by-step installation from the documentation. So far no touch screen, GPIO or anything. Just the pi, a HDMI monitor, keyboard, mouse and camera. In this instance the relevant camera is a Nikon D7000 DSLR.

I've installed the picam, and that works fine. However, I've plugged my Nikon DSLR into the Pi, checked gphoto2 --auto-detect to see it was connected, and tried using it with both commandline and python. Both of them initiate the camera and take photos, exporting them, and all is well. What it doesn't do is autofocus. If I put something close to the lens, or shift it towards an object further away everything is out of focus..

Is there any way to make it do a autofocus before at least the first photo?

reuterbal, 2019-07-08:
Autofocus should automatically be triggered when capturing pictures. Are you sure autofocus is enabled on your lens?

knutta, 2019-07-08:
Yes. I’ve checked it by pushing the release button halfway down manually to see if it focuses. I’ve also double checked both the selection on the lens (Tamron 17-50) and on the camera itself.

reuterbal, 2019-07-08:
Does it focus when you trigger gphoto on the terminal?

knutta, 2019-07-08:
I haven’t actually figured out how to do it myself on the terminal🙈 Do you know what command to issue?

reuterbal, 2019-07-08:
```
gphoto2 --capture-image-and-download
```

knutta, 2019-07-08:
No, no autofocus from the command

knutta, 2019-07-08:
I'll try to see if there's any firmwareupdates to the camera

knutta, 2019-07-08:
Found an update for the firmware, but still no autofocus unfortunately

reuterbal, 2019-07-09:
Just to be sure: Autofocus works when you use your camera normally?

Since gphoto doesn't trigger the autofocus, this is out of my scope. You could try with the gphoto-folks at https://github.com/gphoto/gphoto2. Once Autofocus works when using the gphoto2 command line utility, it should also work with the photobooth.

thyristor59, 2019-07-09:
hello knutta. 

I have a nikon D3200 for autofocus, mine does it when starting the machine, but not between each photo, it does not pose a problem as there is no change of light too important! this does not prevent that if you find a solution I am listening because as usual I am not e 'meusure to do it alone. 

thank you in advance

knutta, 2019-07-09:
It doesn’t autofocus on the first photo, so I guess I’m worse off than you😅 I’ll head over to the gphoto-people and ask there.

thyristor59, 2019-07-09:
was interested in the operation of the device this afternoon. I simply clean the lens because I thought the lens motor was HS. a simple cleaning of the device was enough for me.

I have no problem booting the photobooth

knutta, 2019-07-09:
I found a workaround after searching the gphoto github-issues. It seems if I write the following in command:

`gphoto2 --set-config /main/actions/autofocusdrive=1`

it will focus. However, it doesn't change focus for the next photo. If I write the following command:

`gphoto2 --set-config /main/actions/autofocusdrive=1 --capture-image-and-download`

it will focus, and then shoot and download. Now I only need to find where the command is issued in the photobooth and amend the code. Any tips in where to look? (also - any tips in where I should go to change background to a custom one (for the photobooth) and customize all text?)

Thanks!

thyristor59, 2019-07-09:
to change the order you can search in: frames.py pyqt5gui.py 
If I'm not mistaken! 

your solution interests me if you find solution. 

look in the sheetstyle, it will change the layout. 
for the text you will find a translation file in the folders to correct your text I do not think I'm wrong about the information

reuterbal, 2019-07-09:
For the autofocus configuration, you can create a copy of [photobooth/camera/models/default.cfg](https://github.com/reuterbal/photobooth/blob/master/photobooth/camera/models/defaults.cfg) with your model name in the same folder. Have a look at the photobooth.log output for the model name that is reported by gphoto2 (Or post it here and I'll tell you the file name).

Have a look at the one for the [Canon EOS 500D](https://github.com/reuterbal/photobooth/blob/master/photobooth/camera/models/canoneos500d.cfg) or post the output of your photobooth.log and I'll suggest the file content.

thyristor59, 2019-07-16:
Hello ! d

Do you find a solution for auto focus between each photo? 
I am curious to try! 

Thank you in advance

knutta, 2019-07-16:
I haven’t been able to try yet. Been on vacation for a few days, but will hopefully be able to have a go at it this weekend.

reuterbal, 2019-07-17:
Autofocus can be triggered on most models by setting config values. See the following links for some context:
http://gphoto-software.10949.n7.nabble.com/EOS-600D-Autofocus-Enable-Disable-td13657.html
https://sourceforge.net/p/gphoto/mailman/message/36327009/

knutta, 2019-07-20:
I've made it!

a few hurdles, as I've used the touch screen, and therefore some of the text was beneath the lower edge of the screen. Guess I'll be tackling that next..

My workaround has been to edit the photobooth/camera/CameraGphoto2CommandLine.py to add the 
`--set-config /main/actions/autofocusdrive=1`
right before the
`--capture-image-and-download`

I'm not fluent in python by any means, and not a brilliant programmer in any way, so adding a cfg-file for the camera model was not the way to go - this time. It's only a month till my wedding, so I've got a few other things do to other than making a pretty code before that time...

knutta, 2019-07-20:
By the way - I'm working on the localisation-file as well. But I can't see anywhere to tell the photobooth to use my locale-file... Can't find anything in the manual aswell.. (Am I a bit dense and didn't read properly?)

Any tips in how to make it use my locale-file?

thyristor59, 2019-07-23:
just as knutta I tried to edit. in the commandline folder the autofocus is done successfully, but I can not find a name anymore or edit in cameragphoto 2.py. would you have a solution? thank you in advance

knutta, 2019-07-23:
Here's my command-line-file in case you would like to see it. Haven't ventured into any of the others yet. It's working, so maybe I'll look at them after the wedding ;)
[CameraGphoto2CommandLine.py.zip](https://github.com/reuterbal/photobooth/files/3423504/CameraGphoto2CommandLine.py.zip)

simon3343, 2023-04-06:
Hi. Me and my mate are working on a photobooth using this interesting project. Currently it runs with a Nikon D3200 on a RPi 3 and we also encountered the problem with the missing autofocus. I've tried a few hours this afternoon and got it working for now in the way @knutta did:

> My workaround has been to edit the photobooth/camera/CameraGphoto2CommandLine.py to add the `--set-config /main/actions/autofocusdrive=1` right before the `--capture-image-and-download`
 
After this modification and selecting the gphoto-commandline driver in the photobooth settings, the camera does the autofocus just fine and takes nice pictures, but the liveview doesn't work.....selecting python-gphoto2 it is exactly vice versa, I can see the live view and the camera takes the pictures without problems, but doesn't focus....

I also tried the way described by you, @reuterbal, with the camera configuration files, but I'm a bit clueless about the correct syntax within the file. when I place a file called d3200.cfg in the folder, it seems to be parsed, because if there's any useless or wrong-syntaxed content in it, the software runs into an error, which is fixed when I delete the file and restart the software....

 Can you explain me a bit more about the config files, the following questions are spooking around in my head:

- is the mentionend config file parsed by gphoto or the photobooth program itself?
- does the example file for the Eos 500D contain all available attributes?
- if no, is there a manual, guide or something else like this for the correct usage of these attributes?
- are the attributes all the same for different camera models and manufacturers?

I'm curious about your replies, to all the folks that have already found a solution for this problem, please let us know!

Best regards from germany

simon3343, 2023-05-05:
So we found a solution. After a few weeks frustrating search and useless work we decided to buy a used Canon EOS 500D, which works out-of-the-box, provides Live-View and does the autofocus fine before every single picture. For people who also want to use an EOS 500D, set the program wheel to "Av", Liveview and Autofocus only work in this mode.