reuterbal/photobooth #151: Could not find a version that satisfies the requirement opencv-python

poldixd, 2019-07-10:
**Describe the bug**

When I run the `pip install -e .` command I got this error:

```
  ERROR: Could not find a version that satisfies the requirement opencv-python (from photobooth==0.5.dev3+g40f1cda) (from versions: none)
ERROR: No matching distribution found for opencv-python (from photobooth==0.5.dev3+g40f1cda)

```

**To Reproduce**

Follow the `INSTALL.md` instructions till `pip install -e .` .

**Hardware (please complete the following information)**

 - Device Raspberry Pi 3B
 - Camera Canon EOS 400D
 - GPIO: No

**Software (please complete the following information)**

 - OS Raspian 10
 - Python version 3.7.3
 
 **Installed packages**
 
 ```
arandr==0.1.9
asn1crypto==0.24.0
automationhat==0.1.0
blinker==1.4
blinkt==0.1.2
buttonshim==0.0.2
Cap1xxx==0.1.3
certifi==2018.8.24
chardet==3.0.4
Click==7.0
colorama==0.3.7
colorzero==1.1
configparser==3.5.0b2
cookies==2.2.1
cryptography==2.6.1
drumhat==0.1.0
entrypoints==0.3
enum34==1.1.6
envirophat==1.0.0
ExplorerHAT==0.4.2
Flask==1.0.2
fourletterphat==0.1.0
funcsigs==1.0.2
gpiozero==1.5.1
gyp==0.1
idna==2.6
ipaddress==1.0.17
itsdangerous==0.24
Jinja2==2.10
keyring==17.1.1
keyrings.alt==3.1.1
MarkupSafe==1.1.0
mcpi==0.1.1
microdotphat==0.2.1
mock==2.0.0
mote==0.0.4
motephat==0.0.2
numpy==1.16.2
oauthlib==2.1.0
olefile==0.46
pantilthat==0.0.7
pbr==4.2.0
phatbeat==0.1.1
pianohat==0.1.0
picamera==1.13
picraft==1.0
piglow==1.2.4
pigpio==1.42
Pillow==5.4.1
pycairo==1.16.2
pycrypto==2.6.1
pyflakes==2.0.0
pygame==1.9.4.post1
PyGObject==3.30.4
pyinotify==0.9.6
PyJWT==1.7.0
pyOpenSSL==19.0.0
pyserial==3.4
pyxdg==0.25
rainbowhat==0.1.0
requests==2.21.0
requests-oauthlib==1.0.0
responses==0.9.0
RPi.GPIO==0.6.5
RTIMULib==7.2.1
scrollphat==0.0.7
scrollphathd==1.2.1
SecretStorage==2.3.1
sense-emu==1.1
sense-hat==2.2.0
simplejson==3.16.0
six==1.12.0
skywriter==0.0.7
sn3218==1.2.7
spidev==3.3
touchphat==0.0.1
twython==3.7.0
unicornhathd==0.0.4
urllib3==1.24.1
Werkzeug==0.14.1
 ```

SkiTheSlicer, 2019-07-11:
I mentioned the same issue on the [Raspberry Pi 4 thread](https://github.com/reuterbal/photobooth/issues/142#issuecomment-506899221). Piwheels is still not caught up on building all the pip packages for ARM and Raspbian Buster. Hopefully it's just a matter of waiting it out.

reuterbal, 2019-07-11:
If you want to use the Canon EOS 400D you won't require OpenCV. In that case, you can temporarily remove `opencv-python` from https://github.com/reuterbal/photobooth/blob/40f1cdadcc8cff0c142fe1d4f14813f07c5a656f/setup.py#L186-L193
and re-run the install command.

Hopefully the other packages are already available for Buster. Please report back.

oelegeirnaert, 2019-07-11:
I'll try this solution this evening as well...

poldixd, 2019-07-11:
After I removed `opencv-python` I got this error `photobooth requires Python '>=3.5' but the running Python is 2.7.16`. Python  version 2 (`python`) and 3 (`python3`) is preinstalled on raspbian. I'm a beginner... How can I set version 3 as the "default" python?

reuterbal, 2019-07-12:
I recommend creating a virtual environment (see [this point](https://github.com/reuterbal/photobooth/blob/master/INSTALL.md#initialize-virtualenv) of the installation instructions). That way, you can have multiple versions of the same Python packages on the system.

Good to know that OpenCV is the only package missing on Buster. Maybe we should make that optional as hardly anyone seems to use webcams anyways.

poldixd, 2019-07-12:
> I recommend creating a virtual environment (see [this point](https://github.com/reuterbal/photobooth/blob/master/INSTALL.md#initialize-virtualenv) of the installation instructions). That way, you can have multiple versions of the same Python packages on the system.

ah, ok, I created the virtual environment at the beginning, but I did not know for what it was.

Thanks for your help and for the photobooth software.

closed 2019-07-12 by poldixd