reuterbal/photobooth #153: Photo's Size

alopezbu, 2019-07-16:
Hi, 
I was able to install and run the photobooth, great work btw!. I have built it with a raspberry pi 3b and the picamera. I have activated the option of saving each indepedent picture, appart from the collage. The thing is that the size of the individual pictures is the same as the resolution of the screen. Where can I definde the size of this pictures, or if I cannot choose the size, where can I resize them? As I have said I am using the picamera.

Thank you in advance.

oelegeirnaert, 2019-07-16:
Hi @alopezbu , 
I'm not realy sure and I didn't test it but I guess you should change it over here:
https://github.com/reuterbal/photobooth/blob/master/photobooth/camera/CameraPicamera.py#L72

Kr,
Oele

alopezbu, 2019-07-16:
Hi @oelegeirnaert , 
that was my back up plan, I was thinking to add the line:
 camera.resolution = (1024, 768)
in the same file that you were pointing.
I do not have my raspi with me now but I will try both options tonight and I will see.

Thank you very much.

reuterbal, 2019-07-17:
What version of the Picamera are you using?
The individual pictures should be captured at the native resolution of the camera, as pointed out by @oelegeirnaert 
https://github.com/reuterbal/photobooth/blob/40f1cdadcc8cff0c142fe1d4f14813f07c5a656f/photobooth/camera/CameraPicamera.py#L72
Use something different for resize, e.g., `resize=(1920,1080)` to obtain a different file size.
However, note that the picamera always uses the native resolution for capturing the image, see https://picamera.readthedocs.io/en/release-1.10/fov.html#camera-modes for details.

Elta85, 2019-07-20:
Hi all,

Maybe this works?

Change in photobooth/photobooth/camera/CameraPicamera.py :
`    def getPicture(self):

        self.setActive()
        stream = io.BytesIO()
        self._cap.capture(stream, format='jpeg', resize=None)
        stream.seek(0)
        return Image.open(stream)`

to:

`    def getPicture(self):

        self.setActive()
        self._cap.resolution = (1024, 768)    #line added. You can change the resolution here to something higher then your screen.
        stream = io.BytesIO()
        self._cap.capture(stream, format='jpeg', resize=None)
        stream.seek(0)
        return Image.open(stream)`

This should set the stream to a higher resolution, i think.