reuterbal/photobooth #154: How to add Canon_SELPHY_CP1300.ppd

qwertyFrank, 2019-07-21:
I'm really wondering how to install supplementals/Canon_SELPHY_CP1300.ppd on a RPI via WLAN using gutenprint printer driver.

CP1300 not able to connect to WLAN AP. Therefore I started direct connection on the CP1300. But I have no idea how to add via WLAN within CUPS administration interface.

I still using main branch.

Thanks for any hint.

Elta85, 2019-07-21:
Hi,

Have you checked and tried the instructions at:
https://github.com/reuterbal/photobooth/wiki/Printer-configuration ?

qwertyFrank, 2019-07-22:
Hi, 
Thanks for that hint. I will try.

qwertyFrank, 2019-07-23:
Hi, looks like that CP1300 only able to support 2,4 GHz WLAN. But my Raspberry Pi B+ and installed RaspAP using 5 GHz. Any idea how to switch AP to 2,4 GHz?
Thanks
Frank

qwertyFrank, 2019-07-24:
Hi,

two issues:

First: using ppd from "supplementals" I get error message by CUPS administration, that file is not ok. Using ppd from https://www.voss.earth/2018/08/31/kurztipp-canon-selphy-wlan-drucker-cp910-oder-cp1300-unter-linux-cups-verwenden/, CUPS do not produce error message and I can finish adding the printer, but printing do not work. Get error message in CUPS queue, that CP1300 do not respond even connected via WLAN to RaspAP's AP. Both ppds are completely different.

[Canon_SELPHY_CP1300-(from reuterbal).zip](https://github.com/reuterbal/photobooth/files/3428313/Canon_SELPHY_CP1300-.from.reuterbal.zip)
[Canon_SELPHY_CP1300_(from_link).zip](https://github.com/reuterbal/photobooth/files/3428318/Canon_SELPHY_CP1300_.from_link.zip)

Any hint?

In addition still not figured out how to set AP to 2,4 GHz because CP1300 only working in that mode.

As soon as connected to connect via 5 GHz to photoboof as client (for file transfer, etc.). AP is not any more able to work in 2,4 GHz mode. 

Regards,
Frank

Elta85, 2019-07-24:
Hi Frank,

Got the same issue.
A workaround is to into raspAP webconfig:

1. Go to Configure HotSpot >> Go to the advanced tab.
2. Disable "Wifi client AP mode" >> save settings.
3. Stop the hotspot.
4. Enable the "Wifi client AP mode" >> save settings >> enable hotspot

After this the 2.4 GHz works again (until a reboot).
I guess we have to write a script for this workaround  or check some more logs to find more details about this issue. 

About CUPS:
I don't have the issue with the .ppd that came with the installation (default branch).
I could select the printer after it connected to the raspAP. Then I used the .pdd file that came with the installation as driver.

Hope this helps a bit.

qwertyFrank, 2019-07-25:
Hi Elta85,

How did you connect CP1300 within CUPS? Using ipp, or any other way?

Thanks
Frank

qwertyFrank, 2019-07-26:
Hi,

Finally I got CP1300 connected to the AP. But printing by photobooth not possible (getting stuck after press pring).

Printing by using "lp" command sends jpg to the printer. But CUPS reports that there is no response from the printer (but file is in the printer queue).

Has this line within the ppd-file any impact?
*APSupplies: "http://192.168.1.108:8008/supply/ink_content.html"

Thanks for any hint.

Best regards,
Frank