reuterbal/photobooth #156: {Success story] my wedding photobox

call-me-matt, 2019-07-28:
This is the photobooth that I have built for my wedding. I had programmed a photobooth some years ago with a raspberry pi using the picam, but most of the pictures were blurry. My software was alright, but way less professional compared to this project. Now I wanted to see if I could make it work with my camera and a flash. I spent most of the time in getting my camera tethering to work and enjoyed that a lot. I am very happy with the result, even though it did not work perfectly. Thank you for sharing and maintaining your code, it is terrific!

**Hardware:**
 - Raspberry Pi 3B+ with the Raspi TouchScreen
 - Camera Canon Powershot G5 X
 - No GPIO
 
**Additional equipment**
 - Canon Power Supply from vhbw (Amazon, 14€)
 - Canon Selphy CP910 Photo Printer (used from ebay, 75€)
 - Elinchrome Flash with softbox and skyport radio transmitter (borrowed)
 - Simple tripod screw (2€)
 - Millenium box stands BS-2011 MK II for flash and photobox (2x 24€)
 - KM Box flange 19654 (4€) and ring lock adapter 21441 (borrowed)
 - Face Coasters and Face Mats for fun photos (16€)
 - Wood, screws, angles, vises, wood wax to create the box (90€)
 - Linen as background (2m x 2,80m) (50€)
 - Magnets to attach the background linen (10€)
 - Multiple socket with 5m cable
  
**Modifications:**
 - I created a new theme and also translated the messages to a mix of German, French and English as our wedding was very international
 - I modified the template for printouts to create two photostrips, one of them in black and white as described in #86. Provided scissors so that guests could cut it in the middle.
 - My camera was not supported by gphoto2, so I had to write a driver for that. With extensive support from Marcus Meissner and endless trial and error, I was very happy to discover that in fact tethering was possible - even though the Canon support team answered my request before saying that it would not be. 

**Problems:**
 - As I could not figure out how to wake the camera from idle once in sleep mode, I turned the camera on when people clicked on start and turned it off again in the end. This had the funny effect that the camera lens extended at the beginning and retracted in the end of each picture series.
 - Sometimes, after startup, the program froze just before taking the second picture and crashed with an error message. I had no opportunity to debug - in the beginning I thought it is happening when no internet connection available and a false date leads to overwriting of already existing images, but a manual change of date did not resolve the issue. Anyways, it was not so disturbing, as I could test after setup and the box ran stable afterwards
 - I had some problems with the flash as it turned out to be way too powerful. I should have set the camera to manual mode, but was not thinking about that on the wedding day. Ended up in just turning the steady light on which gave results that are alright.
 - The printer stopped working after a short while. We found out that every time after turning it off and on again, it would print some images - so we did this from time to time. Still, some pictures were never printed. I think this issue is currently being worked on.

**Pictures:**

The first prototype with minimal (cardboard) housing:
[image]

The construction phase of a more robust case:
[image]

The finished work:
[image]

Backside:
[image]

Inside:
[image]

The sketch I made for planning:
[image]

The result
[image]

MatthiasRoelandts, 2019-11-14:
Hello @call-me-matt,
I also would like to buy a Canon Selphy (to use as printer).
Is it possible to print on 2x6 inch paper instead of printing 2 times the 2x6 image on 4x6 paper?
I would like to cut the 4x6 paper in 2 beforehands so it's not needed when the paper comes out the printer

call-me-matt, 2019-11-14:
As far as I know only the regular photo paper exists and the printer does not support half size.