reuterbal/photobooth #162: Nikon D3200

thyristor59, 2019-08-12:
I am looking for a durable solution for starting my Nikon D3200!

 the automatic start works 1/10. the rest of the time I'm struggling to recognize the camera. 
If a person has a solution I am a taker. thank you in advance !
-------------------------------
light the raspberry. 
To be on the program menu.
click on start.
 "photobooth start" without launch.

thyristor59, 2019-08-16:
you can close the subject. I found a solution thank you

katanapod, 2019-08-18:
It would be kind of you if you share your solution...
Don´t let github transform to unusable BulletinBoards

wikijm, 2019-08-18:
And I'll be there to add it in the Wiki 😉

thyristor59, 2019-08-18:
I simply changed the USB cable of the camera, it seemed finally damaged so do not send visibly correctly