reuterbal/photobooth #163: Photobooth does not work on "Buster"

tobiasschuerg, 2019-08-12:
More an fyi than a bug. However, maybe there is a way to disable `open_cv` temporarily as it is probably only used for image manipulation?

## what happen?

`pip install -e .` results in

> Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
> Obtaining file:///home/pi/photobooth
> Collecting opencv-python (from photobooth==0.5.dev3+g40f1cda)
>   ERROR: Could not find a version that satisfies the requirement opencv-python (from photobooth==0.5.dev3+g40f1cda) (from versions: none)
> ERROR: No matching distribution found for opencv-python (from photobooth==0.5.dev3+g40f1cda)

As there are no packages for python 2.x, 3.7 and 4.x. See https://github.com/piwheels/packages/issues/28

So the only *work-around* is to use old `Raspbian Stretch`

> We don't have Python 2 wheels. We also don't have v4.x releases and we don't have any wheels for Python 3.7 (in Buster). If you want to use opencv you'll have to use Raspbian Stretch. There sudo pip3 install opencv-python will work and bring in the latest version we have.

patkan, 2019-08-20:
This is a known problem: #160 and #151

patkan, 2019-09-10:
I have just installed the software on a current buster (RPi 4) and now an opencv package is available.