reuterbal/photobooth #168: ERROR - Camera: [-105] Unknown model

dsheard, 2019-08-24:
2019-08-24 23:01:20,977 - root - INFO - Initializing photobooth
2019-08-24 23:01:20,980 - root - INFO - Photobooth version: 0.5.dev3+g40f1cda
2019-08-24 23:01:20,987 - root - INFO - Reading config file "/home/pi/photobooth/photobooth/defaults.cfg"
2019-08-24 23:01:20,999 - root - INFO - Reading config file "photobooth.cfg"
2019-08-24 23:01:21,091 - root - INFO - Number of last existing file: 0
2019-08-24 23:01:21,099 - root - INFO - Saving pictures as "2019-08-24/photoboothXXXXX.jpg"
2019-08-24 23:01:21,108 - root - INFO - Number of last existing file: 0
2019-08-24 23:01:21,111 - root - INFO - Saving pictures as "2019-08-24/photobooth_shot_XXXXX.jpg"
2019-08-24 23:01:21,111 - root - INFO - GPIO disabled
2019-08-24 23:01:22,215 - root - INFO - Using printer ""
2019-08-24 23:01:25,498 - root - INFO - Using python-gphoto2 bindings
2019-08-24 23:01:26,001 - root - ERROR - CameraProcess: Exception "[-105] Unknown model"
Traceback (most recent call last):
  File "/home/pi/photobooth/photobooth/main.py", line 66, in run
    if cap.run():
  File "/home/pi/photobooth/photobooth/camera/__init__.py", line 94, in run
    self.handleState(state)
  File "/home/pi/photobooth/photobooth/camera/__init__.py", line 101, in handleState
    self.startup()
  File "/home/pi/photobooth/photobooth/camera/__init__.py", line 62, in startup
    self._cap = self._cam()
  File "/home/pi/photobooth/photobooth/camera/CameraGphoto2.py", line 42, in __init__
    self._setupCamera()
  File "/home/pi/photobooth/photobooth/camera/CameraGphoto2.py", line 58, in _setupCamera
    self._cap.init(self._ctxt)
gphoto2.GPhoto2Error: [-105] Unknown model
2019-08-24 23:01:26,018 - root - ERROR - Camera: [-105] Unknown model

MatthiasRoelandts, 2020-02-16:
Is your camera in the list of supported camera's?
[http://gphoto.org/proj/libgphoto2/support.php](http://gphoto.org/proj/libgphoto2/support.php)
try a `ghoto2 --auto-detect` in a terminal

Alabamain, 2021-07-15:
I was getting this as well, and then I remember I never turn my camera on in my raspberry pi. If you have a raspberry pi and get this error go and make sure it is on.