reuterbal/photobooth #170: One shot

thyristor59, 2019-09-02:
Hello, 
I participated in a wedding this weekend, there was a professional photobooth that allowed to choose only 1 shot or x4. 

I would like to do the same for my 30 years. I can not do this.
 if anyone has a track or solution I would be very happy! 
some pictures:
[image]

patkan, 2019-09-02:
This is a duplicate of #158 ;-)

Tinker232, 2019-11-14:
Hi @patkan 
I guess the #158 won´t help for this issue so far?!
Are there anybody who has an idea, how to realize a additional "layer" after pressing the button to choose between various picture layouts?
LG Tina

patkan, 2019-11-14:
Hi @Tinker232, nope #158 hasn't any implementation yet. As far as I see it, there are two options:
* having different buttons like in #159 and sending different events
* if you want to have something like an extra dialogue, you would have to add an extra state together with a GUI-frame. This extra state could then make available all of the options.