reuterbal/photobooth #171: Canon EOS450D: configuration file seems mandatory

andyboeh, 2019-09-03:
**Describe the bug**

I'm building a Photobooth based on a Canon EOS450D camera. It took me quite some time to figure out that the camera only focuses before taking a shot if a camera specific configuration file exists - for whatever reason!
I simply copied the canoneos500d.cfg to canoneos450d.cfg and that's it - the camera now focuses.

**To Reproduce**

Steps to reproduce the behavior:
1. Take pictures with a Canon EOS450D - the camera will focus only during initialization and never again.

**Expected behavior**

The camera should focus before taking a picture.

**Hardware (please complete the following information)**

  - Raspberry Pi 3B
  - Canon EOS 450D
  - GPIO: No
  - 7" Raspberry Pi Official Touchscreen

**Software (please complete the following information)**

 - Raspbian Buster
 - Python 3.7.3

wikijm, 2019-09-03:
Hi @andyboeh,

Thanks for your feedback 😃
Could you please give us the content and the path of the file you are talking about?

Sincerely,
WikiJM

andyboeh, 2019-09-03:
Of course, to be more specific: photobooth/camera/models/canoneos500d.cfg needs to be copied to photobooth/camera/models/canoneos450d.cfg. The content can stay the same: [canoneos500d.cfg](https://github.com/reuterbal/photobooth/blob/master/photobooth/camera/models/canoneos500d.cfg).