reuterbal/photobooth #172: External flash does not fire

pedakas, 2019-09-04:
Hello,
I have a small problem, everything works fine except that an external flash (Walimex Newcomer 100) via a synchro cable doesn't trigger with Python-Gphoto2, the integrated flash works and if the Photobooth doesn't run the external flash also works. With gphoto2-commandline the external flash also works. I think I remember that it worked in an earlier version. Anyone has an idea? Thanks

Hallo,
ich habe ein kleines Problem, es funktioniert alles wunderbar bis auf das ein Externer Blitz (Walimex Newcomer 100) über ein Synchro Kabel nicht auslöst mit Python-Gphoto2, der Integrierte Blitz funktioniert und wenn die Photobooth nicht rennt funktioniert auch der externe Blitz. Mit gphoto2-commandline funktioniert auch der externe Blitz. Ich glaube mich zu erinnern das es in einer früheren Version funktioniert hat. Jemand eine Idee? Danke

 - Device  Raspberry Pi 3B+
 - Camera Canon EOS 550D]
 - GPIO: No

andyboeh, 2019-09-04:
Just a guess, but it might be that the EOS550D also needs a configuration file, similar to my EOS450D. See #171 for details.

pedakas, 2019-09-05:
> Just a guess, but it might be that the EOS550D also needs a configuration file, similar to my EOS450D. See #171 for details.

Hi @andyboeh Thanks, i will try this in the evening.

pedakas, 2019-09-06:
Hi @andyboeh Your tip worked, the flash is now working fine! Thank You!

closed 2019-09-06 by pedakas