reuterbal/photobooth #175: GPIO for REPEAT and PRINT + Flashing LED for Printing Process

loeckchendasboesschen, 2019-09-08:
Hello everybody,

i started building my Photobooth with reuterbals software (really really cool by the way!).
Everything is working so far (Touchscreen, Canon 1300 Printer,  Preview, Printing, Button for Trigger and Flashing LED during Countdown).
I'm using one of those Arcade Buttons with an build in LED. So I thought about 3 buttons in a row, to trigger all events instead of using the touch screen.
Has anybody here done this already and could help me out with some code? I tried to understand the gpio.py but couldn't get much out of it.

gpsless, 2019-09-18:
@reuterbal, I second @	loeckchendasboesschen 's request.
I personally suggest that after postprocessing the "start over" command should be triggered by the trigger button, the "print" command via a separate GPIO.
I have a friend who is much more acquainted to Python than me, and he is helping me in modding the code.
Could you please advise us where to insert the GPIO event for a print button please?

P.S.: I will use the photobooth for my wedding in less than two weeks, if you could help me make the photobooth "foolproof" (in other words "drunken-proof"), it would be really great! :)

Many thanksè

gpsless, 2019-09-18:
> i started building my Photobooth with reuterbals software (really really cool by the way!).
> Everything is working so far (Touchscreen, Canon 1300 Printer, Preview, Printing, Button for Trigger and Flashing LED during Countdown).

@loeckchendasboesschen, which GPIO exactly triggers the flashing effect for countdown?
Is it a standard feature?
If not, could you forward me the modded files please? :)

Many thanks!