reuterbal/photobooth #180: Compact Printing photobooth

andyboeh, 2019-09-15:
**Describe your photobooth**

I built the box for weddings and parties - the first successful use was at a friend's party. I wanted my box to be very compact and all-in-one, so I came up with a 40cm x 30cm x 30cm wooden case.

**Hardware (please complete the following information):**
 - Raspberry Pi 3B + RTC module
 - 7" Official Raspberry Pi Touchscreen
 - Canon EOS 450D
 - Neewer NW670 TTL Flash + Extension Cord
 - GPIO: No
 - Canon Selphy CP1300

**Software (please complete the following information):**
 - Raspbian Buster

**Modifications:**
The Raspberry Pi rootfs was switched to read-only mode and I added an overlayfs for read-write. In order to store pictures, I added a python script that checks for a USB drive on startup and, if found, mounts it. I then modifies the photobooth configuration and starts photobooth. This has the nice side-effect that it also acts as a splash screen.

**Problems:**
 - For no obvious reason, the power supply of the camera died after two hours of use. Fortunately, it's a 5V 3A USB supply and the 5V 2A supply of a party guest worked just as well.
 - The printer is connected via WiFi and sometimes the printer stopped printing (happend twice). I connected via my cell phone and restarted CUPS, then printing resumed. I am going to add a script to automate restarting CUPS if a job stalls.
 - I noticed that the printer compartment gets rather warm. I will add a fan.
 - Very rarely the flash didn't fire. I suppose that the power supply is a bit too weak (5V 4A supply I had lying around).

**Pictures:**
[image]
Camera is mounted on the bottom, using a screw on the bottom side, it can be tilted.

[image]
Canon Printer

[image]
Canon Printer

[image]
The flash was not modified, but an AA battery adapter was adapter is included. I had the power supply lying around, it supplies 4A at 5V.

[image]
Completed box

[image]
Printer at the side of the box

[image]
Printer is accessible for replacing cartridges

[image]
Printer is accessible for replacing cartridges

thyristor59, 2019-09-16:
hello, very nice box !!!
 having to manually restart CUPS every time I change ink and / or paper. 
I  would be curious to see the scrips to no longer have to do it manually! would be nice

andyboeh, 2019-09-17:
@thyristor59 I'll post/link the scripts here as soon as they are ready - might take a few days.

andyboeh, 2019-09-21:
@thyristor59 I tried to reproduce the CUPS problem at home, but I don't want to waste much photo paper. So far, I wasn't able to get cups in a state where I wouldn't continue on its own. Do you know how to trigger this behavior? Running out of paper is not enough in mine case, it continues fine after paper has been added.

oelegeirnaert, 2019-09-24:
Hi @andyboeh ,

What's the name of the fabric you used to cover the wooden case?

Kind regards,
Oele Geirnaert

andyboeh, 2019-09-24:
That's felt used for building speakers - I used some leftovers from the subwoofers I built a while back. I used spray adhesive, but you can buy the felt with adhesive on.

The red door is covered using a self-adhesive foil.

closed 2019-10-10 by andyboeh