reuterbal/photobooth #181: Display black during installation / Display schwarz bei der Installation

MarioJanssen27, 2019-09-18:
https://github.com/reuterbal/photobooth/blob/master/INSTALL.md
Image: Raspbian Buster with desktop and recommended software

The display remains black after installation. At each new start.
What do I have to change?

------------------------------------------------------
in deutsch
https://github.com/reuterbal/photobooth/blob/master/INSTALL.md
Image: Raspbian Buster with desktop and recommended software

Der Display bleibt nach der Installation schwarz. Bei jeden neuen Start.
Was muss ich ändern?

andyboeh, 2019-09-18:
Do you get any error message? 
What does the log file say (photobooth.log)?
When exactly is the display turning black? I can see the desktop for a few seconds until photobooth is started.

MarioJanssen27, 2019-09-18:
> Do you get any error message?
> What does the log file say (photobooth.log)?
> When exactly is the display turning black? I can see the desktop for a few seconds until photobooth is started.

Do you get any error message? **NO**
What does the log file say (photobooth.log)? **Screen at start black**
When exactly is the display turning black? I can see the desktop for a few seconds until photobooth is started.
Photobooth or Desktop do not start. Only RasperryPi startup screen and then it is black.

andyboeh, 2019-09-18:
Then it's probably best to consult the system log if the desktop environment doesn't come up - can you log in via SSH or switch to a console? You can use `sudo journalctl -b` to get the log from the current boot and, with some other options, from any previous boot.

MarioJanssen27, 2019-09-18:
> Then it's probably best to consult the system log if the desktop environment doesn't come up - can you log in via SSH or switch to a console? You can use sudo journalctl -b to get the log from the current boot and, with some other options, from any previous boot.
**I can't log in to the console**

MarioJanssen27, 2019-09-18:
> Then it's probably best to consult the system log if the desktop environment doesn't come up - can you log in via SSH or switch to a console? You can use sudo journalctl -b to get the log from the current boot and, with some other options, from any previous boot.
**I can't log in to the console**

I just tried a new installation over pibackery.
With the following message it did not go on:
SSH is enabled and the default password for the 'pi' user has not been changed. This is a security risk - please login as the 'pi' user and type 'passwd' to set a new password.

But my backup doesn't work with a new installation of the new version of Photobooth :-(