reuterbal/photobooth #184: Feature Request / Question: Relay for Camera

andyboeh, 2019-09-24:
While testing my first photobooth at a party, I noticed a camera freeze (an infamous Err 99 from the Canon camera) and I had to reboot the whole box to get it working again (it happened once in about 200 photos taken).

In order to automate / shorten the process, I was thinking about adding a relay to the camera power supply that is controllable via the Raspberry's GPIO so that the software is able to power-cycle the camera.

Then, the software needs to be improved with regard to error handling: If a severe fault occurs, it should give the option to retry by power-cycling the camera and performing the last action (e.g. taking a picture) again.

- Has any work being done into this direction?
- Any comments or suggestions on this?
- Does anybody know of a way to automatically power on the Selphy CP 1300?

I have yet to order a relay module, so it's going to take some time until I can start working on this.

botyas, 2019-09-24:
to turn on the printer I soldered a cable between the pins of the power button and activated it with a 5v relay on a pin of the Raspberry gpio and with a cron task it triggered it 1 second at startup.  I use the printer exclusively for photobooth

botyas, 2019-09-24:
I have used a Canon Eos 450D camera with gphoto and Logitech C901 opencv and with the 2 I have taken more than 200 photos without problems.  the 450d works with a dummy battery connected to 5v of the atx power supply that I have to power all the devices, led strip, router, flash, Raspberry and others

andyboeh, 2019-09-25:
@botyas thanks for your comment! I've got an EOS 450D as well, it's also powered via a DC coupler. It came with a 5V 3A supply that died after two hours of use. A 2A supply leads to frequent Err 99, so I suspect that power is the issue here.
Can you power the Selphy via the ATX supply as well (it needs 18V, right?)?

websta911, 2019-11-19:
Hello, 
If you still have the issue of the Photobooth freezing after some idling time, please check if there is a config file in  photobooth/camer/model/  should be something like "canoneos450d.cfg"  if you check the photobooth.log you should see a line during startup that reads something like loading config  and there you should find the correct name for the file. 

Easiest solution just copy the file, that is already in there and rename the copy according to your needed config file name. That should fix the the freezing issue.

best regards

andyboeh, 2019-11-19:
@websta911 thanks for your comment, but I get an Err99 from/on the camera - this happens immediately after taking a picture, not during idling. I am well aware of the missing configuration file, given the fact I was the one to report it (#171 and #191).