reuterbal/photobooth #186: [Build Story] New Compact Printing Photobooth

andyboeh, 2019-09-30:
**Describe your photobooth**
This is going to be my second photobooth and I decided to post early on, during the development process, to get some feedback and ideas. 
As this is the initial post, this is what I've planned and partially ordered so far:

**Hardware (please complete the following information):**
 - Raspberry Pi 4, 1GB RAM
 - Canon EOS 1100D (used, broken screen) + EF-S 18-55 IS STM (if I can fix it, depends on spare part availability)
 - Canon Selphy CP1300
 - 7" Touchscreen
 
 **Additional equipment**
 - ATX Power Supply
 - 24V DC booster
 - Neewer NW-670 Flash
 - Small LED lights for AF assist light
 - Arduino Pro Micro for power management
 - 12cm Fan
 - Relay module

 **Planned Build**
I'm planning on building a compact wooden box, similar to my first one (see #180), again with a slide-out printer. This time, I will use an ATX power supply to power the whole system:
 - Raspberry from 5V
 - Camera from 5V
 - Flash from 5V
 - Touchscreen from 12V
 - Arduino Pro Micro from 5V Stand-By
 - Printer via DC-Booster from 12V
 - AF Assist lights from 12V

 **Arduino Pro Micro**
I am planning on integrating an Arduino Pro Micro as supervisor: It will be powered by +5V Stand-By of the ATX supply and thus be available as soon as the system is plugged in. The Pro Micro will:
 - Monitor the button state of the power button
 - Blink the front button light while the system is booting
 - Monitor the state of the Raspberry
 - Shutdown the Raspberry when the front button is pressed (blink the LED again during shutdown)
 - Switch on the printer (using a Relay, I will have to modify the printer for this)
 - Switch the camera and the flash on/off
 - Switch the AF assist light on/off

Of course, the Raspberry and the Arduino will have to communicate, I'll probably go via the UART and implement a simple communication protocol. 

 **Printer**
I would like to get better status feedback from the printer. I've read that the printer reports its status via USB, so I'll probably implement a stand-alone program that monitors the printer via USB while printing actually works via WiFi since it's more reliable for most. Then the photobooth can show printer messages like out of paper or out of ink.
The printer will be modified so that I can power it on/off via a relay. 

  **External Screen**
The Raspberry Pi 4 has two HDMI outputs and is fully dual-screen capable. I'm planning on adding an HDMI socket to the box, so that an external screen (like a big TV screen or a projector) can be plugged in. I would like to extend the software to support a slideshow on the external screen.

 **Status**
I've ordered most components (apart from the box) and I still had some parts, like the Arduino Pro Micro, lying around. This is going to be a spare time project, so don't expect update to happen frequently. Plus, there is quite some work to do regarding the software and hardware implementation.

I'd love to read your comments!

andyboeh, 2019-10-07:
## First update

Last weekend I had quite some time to work on the new project. What I did:
 - Modified ATX power supply to only bring out the relevant lines (5V, 12V, GND, PSON, 5VSB)
 - Connected Arduino Pro Micro to relay module, built small resistor-transistor circuit to switch on the power supply
 - Connected nice button with LEDs to Aruino
 - Started programming the Pro Micro - the infrastructure is in place and can be found at https://github.com/andyboeh/photobooth-psu
 - Started hacking on the printer (software only)

## Printer

As mentioned, I'd like to get better feedback on the printer state in software, so I played around with the different connection methods, drivers and tools. My original idea was to connect it via WiFi for printing and to use the USB connection for status reporting. It doesn't work: While a client is connected, the USB connection is disabled, hence no status output.
I'm currently investigating two solutions:
 - Use IPP to get the printer status report - I haven't tested that and I don't know yet what the printer really reports
 - Use CPNP to get the printer status report - this works, but the protocol is not known in detail to report all printer errors.

Since IPP seems to be the easier solution, i will investigate that first.

A few pictures:
[image]
[image]

andyboeh, 2019-11-25:
# Second Update

It's been some time since I had the time to work on the Photobooth. I built the wooden case including the camera mount and a side-door for easy access to the components. The printer door is still missing and so is the printer mount, therefore, pictures will be added later on.

The next update will probably be made after Christmas, in the hope that there will be some time to work on the project again.

## Printer

In the meantime, I tested the printer connection using IPP over WiFi: The reports from the printer are actually complete, so what's missing is a python module to report the printer status to the photobooth software. I will connect the printer using USB and see if it is really unstable.

## External Screen

Finally, I was able to figure out a configuration on the Raspberry Pi 4 that allows the use of a 7" touch panel along with an external screen. The 7" screen I ordered is a cheap Chinese IPS-screen connected via HDMI with remarkably good value for the price. However, the provided script for adding it to the system is complete unnecessary crap. Only downside is that the Pi does not support hotplug, so all screen have to be connected before powering on the system.

andyboeh, 2020-02-10:
# Hardware (nearly) finished

Finally, an update. The hardware is nearly finished and I'll make a few software adaptations. Regarding hardware, the modification of the printer is not yet done.

## Outside

The whole box is covered with black and red matte self-adhesive foil. When attached properly, it looks really nice. Unfortunately, there are a few bubbles at the front that I couldn't get rid of.
Anyhow, from top to bottom: Flash + assist lights (one red, one white), 7" touch screen, camera opening, power button. The power button is a push button with blue LED light.

[image]

On the bottom, there are USB, HDMI and Ethernet ports.

[image]

The back side contains the system fan, one side is the service door and the other side contains power supply and printer.

## Inside
There is now more hardware (and cables!) included than I had anticipated. I'm not particularly happy with how the cable management is done (mostly, hot glue), but it's OK for now.

Complete inside view, from top to bottom: Flash with thick spiral cable, system fan, printer, on the right hand side display and Raspberry Pi 4, camera, power supply, relay board, system controller. Not visible: MOSFET PWM driver board.

[image]

## Software

There is a small ATMega 32u4 (an Arduino Pro Micro) that runs from the 5V standby line of the power supply. It is responsible for powering on and shutting down the system. Initially, i wanted a bi-directional communication between photobooth and the MCU via serial, but it turned out that I have the 5V version while the Raspberry is 3.3V. Thus, i ended up having a GPIO-based communication, whereby the ATMega only ever pulls pins low or leaves them floating (pull-ups on the Raspberry side). I enabled two DT-overlays on the raspberry side: `gpio-poweroff` and `gpio-shutdown`. The Raspberry then signals via a GPIO line when shutdown has completed and can be shutdown by pulling a GPIO line low. Then, I added a script to autostart that pulls a GPIO low to signal that boot has completed.

The ATMega also blinks and pulses the power LED and powers on/off the camera and the printer (coming soon).

andyboeh, 2020-02-27:
Regarding the printer: It works very well over USB as long as you don't run out of paper or ribbon. If you do, CUPS is paused and upon resume, the job is duplicated. You can't force-resume in a script because you end up with one additional copy for every resume.

I dove into the driver code and I'll try a few adjustments so that CUPS doesn't pause but reports the printer state only. This is, once again, going to take some time...

thyristor59, 2020-03-13:
hello i would be fully interested in solving the problem of pausing the printer! Did you find a lead? thank you in advance

andyboeh, 2020-03-13:
Yes, the lead is as mentioned: the printer driver reports an error state instead of a "warning" if it runs out of paper. I'm not following this direction anymore, because I'm currently re-writing photobooth in C++ and I'll directly include support for the printer driver there.

thyristor59, 2020-03-13:
good luck on the job! I am new, I try to find a solution for this problem via the forums.

andyboeh, 2020-03-13:
If you refer to your problem in #209, that's something entirely different. My printer is connected via USB and uses the Gutenprint backend. Yours is connected via WiFi and uses a generic IPP interface.

thyristor59, 2020-03-13:
these are actually 2 different problems, and I still haven't managed to connect to WiFi

andyboeh, 2020-03-14:
Regarding WiFi: I followed the official tutorial at https://www.raspberrypi.org/documentation/configuration/wireless/access-point.md, works perfectly on Buster. The only thing I added in addition is a dhcp-host entry for dnsmasq with the printer's MAC address so that it always gets the same IP address assigned.

andyboeh, 2020-05-18:
In the meantime, I mixed-in a 3D-printed part (camera holder, so that the camera can be moved forward and backward easily).

Regarding printer support: In the end, I completely rewrote photobooth in C++ and directly included printer support for Canon Selphy via WiFi and USB. Since it no longer runs photobooth, I'm closing this.

closed 2020-05-18 by andyboeh