reuterbal/photobooth #189: Background Image on top

pedakas, 2019-10-10:
Hello,
Is it possible in the picture creation process that the background image is on top, so that i can add flowers or something else like in this picture for example: https://sdeweddings.com/torontoweddingvideophoto/2013/09/sdeweddings-toronto-photo-booth-rental-9.jpg

Thanks!

pedakas, 2019-10-11:
> I suppose #176 is what you are looking for.

Thanks, so if i unterstand it correct i just replace this 3 files with the files from patkan and i should be able to add a "watermark" in the settings menu?

patkan, 2019-10-11:
Yep, that should be it. Alternatively you could use directly the branch at https://github.com/patkan/photobooth/tree/feature-assemble-overlay
In case that you find any issues with it, it would be great, if you could post them in #176

pedakas, 2019-10-14:
Thanks, I will try this next weekend and let you know if it worked