reuterbal/photobooth #194: Photobooth doesn't start

mfausin, 2019-10-16:
**Describe the bug**

I installed the actual version of Photobooth. I can start the Programm with the command "python -m photobooth" and I can do all the settings. But wenn I hit the Button to start the photobooth it does nothing happen. I only can press ESC to get back to the start screen.

I don´t have an idea what's the problem, the log file not really helps me.

im using a Canon EOS450D, I just renamed the originale cfg file "canoneos500d.cfg" to "canoneos450d.cfg", is this right?

**Hardware (please complete the following information)**

 - Device: Raspberry Pi 3B+
 - Camera: Canon EOS 450D

**Software (please complete the following information)**

 - OS: Raspbian Stretch
 - Python version 3.5.3

**Photobooth log**

Run the application as `.venv/bin/python -m photobooth --debug` and paste the logfile here:

```
2019-10-17 00:23:13,679 - root - INFO - Initializing photobooth
2019-10-17 00:23:13,681 - root - INFO - Photobooth version: unknown
2019-10-17 00:23:13,684 - root - INFO - Reading config file "/home/pi/photobooth/photobooth/defaults.cfg"
2019-10-17 00:23:13,690 - root - INFO - Reading config file "photobooth.cfg"
2019-10-17 00:23:13,719 - root - DEBUG - Context: New state is "WelcomeState"
2019-10-17 00:23:13,744 - root - DEBUG - GuiProcess: Initializing...
2019-10-17 00:23:13,744 - root - DEBUG - CameraProcess: Initializing...
2019-10-17 00:23:13,750 - root - DEBUG - WorkerProcess: Initializing...
2019-10-17 00:23:13,753 - root - DEBUG - WorkerProcess: Running...
2019-10-17 00:23:13,758 - root - INFO - Number of last existing file: 0
2019-10-17 00:23:13,761 - root - INFO - Saving pictures as "2019-10-17/photoboothXXXXX.jpg"
2019-10-17 00:23:13,764 - root - INFO - Number of last existing file: 0
2019-10-17 00:23:13,765 - root - INFO - Saving pictures as "2019-10-17/photobooth_shot_XXXXX.jpg"
2019-10-17 00:23:13,780 - root - DEBUG - GpioProcess: Initializing...
2019-10-17 00:23:13,782 - root - DEBUG - GpioProcess: Running...
2019-10-17 00:23:13,784 - root - INFO - GPIO disabled
2019-10-17 00:23:14,226 - root - DEBUG - GuiProcess: Running...
2019-10-17 00:23:15,352 - root - INFO - Using printer "Canon_SELPHY_CP1300"
2019-10-17 00:23:20,326 - root - DEBUG - Context: Handling event "start"
2019-10-17 00:23:20,328 - root - DEBUG - Context: New state is "StartupState"
2019-10-17 00:23:44,147 - root - DEBUG - Context: Handling event "Teardown(2)"
2019-10-17 00:23:44,148 - root - DEBUG - Context: New state is "TeardownState"
2019-10-17 00:23:44,152 - root - DEBUG - Context: Handling event "welcome"
2019-10-17 00:23:44,152 - root - DEBUG - Context: New state is "WelcomeState"
```

andyboeh, 2019-10-18:
Which backend are you using?
Does the dummy backend work?

mfausin, 2019-10-24:
hmm, i don't know what happened. But now it works... 
It seems it just took a long time for the fist time to start. I was waiting 5 minutes or more, then the camera was reacting and the photobooth started. Now when I restart it works absolut normal.

closed 2019-10-24 by mfausin