reuterbal/photobooth #196: Camera and GPIO error after a while

Tinker232, 2019-10-22:
Hey,
I use the modified version from websta911 with fancy layouts etc. with my raspberrypi4 and a canon selphy1300 (connected via wifi).
The photobooth works fine, but when the photobooth is not used for approximately 30/40 minutes and someone hit the button, there is no live picture from the camera and short time later an error occurs. (Camera and GPIO)

**Screenshots/Screencast**

[URL=https://www.bilder-upload.eu/bild-b01312-1571784171.png.html][IMG]https://www.bilder-upload.eu/thumb/b01312-1571784171.png[/IMG][/URL]

**Hardware (please complete the following information)**

 - Raspberry PI4 
 - Camera 70D
 - Canon Selphy CP1300
 - GPIO: [No]

**Photobooth log**

```
2019-10-15 21:16:51,997 - root - INFO - Initializing photobooth
2019-10-15 21:16:51,997 - root - INFO - Photobooth version: 0.5.dev6+g871b526
2019-10-15 21:16:51,998 - root - INFO - Reading config file "/home/pi/photobooth/photobooth/defaults.cfg"
2019-10-15 21:16:52,001 - root - INFO - Reading config file "photobooth.cfg"
2019-10-15 21:16:52,051 - root - INFO - GPIO disabled
2019-10-15 21:16:52,069 - root - INFO - Number of last existing file: 35
2019-10-15 21:16:52,074 - root - INFO - Saving pictures as "2019-10-15/photoboothXXXXX.jpg"
2019-10-15 21:16:52,081 - root - INFO - Number of last existing file: 0
2019-10-15 21:16:52,083 - root - INFO - Saving pictures as "2019-10-15/src/photobooth_shot_XXXXX.jpg"
2019-10-15 21:16:52,087 - root - INFO - fancy "False"
2019-10-15 21:16:52,843 - root - INFO - Using printer "Canon_SELPHY_CP1300"
2019-10-15 21:16:54,117 - root - INFO - Using python-gphoto2 bindings
2019-10-15 21:16:54,403 - root - INFO - Camera summary: Manufacturer: Canon Inc.
Model: Canon EOS 70D
  Version: 3-1.1.1
  Serial Number: f1266c86fd7c41cf848c8b4d6f39d057
Vendor Extension ID: 0xb (1.0)

Capture Formats: JPEG
Display Formats: Association/Directory, Script, DPOF, MS AVI, MS Wave, JPEG, CRW, Unknown(b103), Unknown(bf02), Defined Type, Unknown(b104), Unknown(b105)

Device Capabilities:
	File Download, File Deletion, File Upload
	No Image Capture, No Open Capture, Canon EOS Capture, Canon EOS Capture 2
	Canon Wifi support

Storage Devices Summary:
store_00020001:
	StorageDescription: SD
	VolumeLabel: None
	Storage Type: Removable RAM (memory card)
	Filesystemtype: Digital Camera Layout (DCIM)
	Access Capability: Read-Write
	Maximum Capability: 31037849600 (29600 MB)
	Free Space (Bytes): 20006240256 (19079 MB)
	Free Space (Images): -1

Device Property Summary:
Model ID(0xd049):(read only) (type=0x6) 2147484453
Property 0xd402:(read only) (type=0xffff) 'Canon EOS 70D'
Property 0xd407:(read only) (type=0x6) 1
Property 0xd406:(readwrite) (type=0xffff) 'Unknown Initiator'
Property 0xd303:(read only) (type=0x2) 1
Battery Level(0x5001):(read only) (type=0x2) Enumeration [100,0,75,0,50] value: 100% (100)

2019-10-15 21:16:54,614 - root - INFO - Loading camera config "canoneos70d.cfg"
2019-10-15 21:16:54,638 - root - INFO - Camera configuration:
Camera Actions [actions]: 
    Synchronize camera date and time with PC [syncdatetimeutc]: 0
    Synchronize camera date and time with PC [syncdatetime]: 0
    UI Lock [uilock]: 2
    Drive Canon DSLR Autofocus [autofocusdrive]: 0
[... 221 lines cut ...]
    Auto Exposure Bracketing [aeb]: off (off, )
Other PTP Device Properties [other]: 
    Model ID [d049]: -2147482843
    PTP Property 0xd402 [d402]: Canon EOS 70D
    PTP Property 0xd407 [d407]: 1
    PTP Property 0xd406 [d406]: Unknown Initiator
    PTP Property 0xd303 [d303]: 1
    Battery Level [5001]: 100 (100, 0, 75, 0, 50, )

2019-10-15 23:14:34,034 - root - INFO - Using camera with preview functionality
2019-10-15 23:14:36,122 - root - INFO - _skip: "[4]"
2019-10-15 23:14:36,129 - root - INFO - Adding Logo : "False"
2019-10-15 23:14:36,129 - root - INFO - Using logo "/home/pi/Bilder/1024px-logo.png"
2019-10-15 23:14:36,129 - root - INFO - Using background "/home/pi/photobooth/supplementals/backgrounds/Fotopaper_background_forever_59.jpg"
2019-10-15 23:15:03,754 - root - INFO - Pic "0"
2019-10-15 23:15:03,833 - root - INFO - Pic "1"
2019-10-15 23:15:03,906 - root - INFO - Pic "2"
2019-10-15 23:15:04,077 - root - INFO - Saving picture as 2019-10-15/photobooth00037.jpg
2019-10-15 23:35:57,121 - root - INFO - Exiting photobooth with status code 0

```

**Additional context**

Add any other context about the problem here.

websta911, 2019-11-03:
Hello, I looked into that and I have a hunch.
I looked into the new camera/gphoto2 module since there were some changes.
I am not exactly sure what is happening precisely but I copied over an old version and it ran with my eos 550 but not with eos m3. Whats the difference:
in the "old" implementation there are some settings which are beeing set when changing from active to Idle. If you do not have a config file with those settings the new module does not do it so you stay active all the time and I am not sure if that is the problem but I think it is. Because with the old version and the eos 550 the box could sit on "press the button" for some time and not freeze.

In the old module during idle the output is set to Off. that does not happen when you do not have a cfg file for your camera model under camera/models/

Your configuration looks fine so you can try copying the eos500d.cfg file already in the models folder and paste it as canoneos70d.cfg and try it out. If during startup or during running you are getting some error you can try deleting the problem cfg line or refine it using the cameraconfig in the photobooth log. 

For me I am not sure what to do because my output value is 'unknown value 0008' or 'Off' and that is all that is possible. I have to try to switch between those to but my time this up so I have to try again  another time. I hope there is some sense in that an it works for you if you create a canoneos70d.cfg file.Regards 
Christoph

Tinker232, 2019-11-05:
Hey Christoph,
thanks for your reply. 
Meanwhile I tried the steps from your first post. Trying to control the camera via gphoto2 / python after the error occured also didnt enable the connection.
Copying the config-file and renaming it to canoneos70d.cfg was the solution! Now it works fine, also for a long time in idle without any errros. 
Thank you for your support Christoph...
Regards Tina

closed 2019-11-17 by Tinker232