reuterbal/photobooth #213: [Feature] Flexible templates for assembling pictures

t0mmo, 2020-03-29:
Inspired by #148 (@websta911 very nice layouts!), #156, #176, #189, #86 and other requests (and my own needs), I added a new templating mechanisms that allows simple configuration of layout instead of writing python code: Simply add an XML file describing where to put the photos (specifying size, position and rotation) and point photobooth to it.

All changes are still in my fork in the freature branch [fancytemplates](https://github.com/t0mmo/photobooth/tree/fancytemplates).

Curious to get your views and feedback - it's still beta and wip

...and thanks to all who have contributed to that nice piece of software so far!

TheSuperpapi, 2020-05-11:
Hello,
Could you explain how to install it?

t0mmo, 2020-05-11:
Hi TheSuperpapi, 
if you checkout this branch, it should just work:
https://github.com/t0mmo/photobooth/tree/fancytemplates
Have fun!

thyristor59, 2020-08-29:
good evening tommo. 
I am a pure novice despite all the investigations that I was able to undertake thanks to reuterbal!
 I would like a very simple way to have your typical real photo booth format. 2 * 4 photos. 
I can't seem to include your download .... 
how should I put take !? 
Thank you in advance !

thyristor59, 2020-09-02:
hello tommo! 
having a little time, I took your version for fun! 
she is awesome ! 
however how can I do so that the final result turns 90 ° during the final poster before clicking on print or start over? 
Thank you in advance

thyristor59, 2020-09-02:
and where can I change the text to be in French? still a novice, I try to improve myself = D

TheSuperpapi, 2020-09-02:
Hello, es tu français ?
Hi, are you a french Guy ?

thyristor59, 2020-09-02:
Yes!
And i handerstand, all IS whrite for me!

t0mmo, 2020-09-02:
Hi thyristor59,
great that you managed to get it working and like the templates.
To change the text you just need to edit the file is the template folder - either the foreground pic, background pic or the xml file - depending on your template design. You find the files in the `supplement/templates` folder.
The 90° turn is a good idea. Need to look into it when time allows.
Curious to see what designs you come up with! maybe you can share your results here...

thyristor59, 2020-09-02:
I take the photobooth again 
I would like a 100% final version, for the moment 
I am trying to find a solution for the white edges that I cannot remove. 
visually on the screen the rotations and the arrangement of the photos are correct. 
Ishare once the work is done = D

thyristor59, 2020-09-02:
for the 90 ° turn I simply modify the xml, and I created a background with French text for modifying the photo background! 
unfortunately, I still have white bors on each side once the photo is printed (selphy cp1300)

t0mmo, 2020-09-02:
hmm... I never had white borders. Maybe check the printer settings? I remember some setting in the printer setup in CUPS which was something with borderless or fit-to-page or similar. Sorry, I don't have my booth readily availible to look it up.

thyristor59, 2020-09-02:
I always had this white stripe problem. 

it is true that with your basic configuration it does not have any. 

at the time I modified the file for the good of my photobooth version, I don't know how to adjust.

 borderless is enabled in cups but without success. 

here is the difference: 
[image]
[image]

it's late in France, I wish you a good day and if you have any ideas I would be very happy! I think it's the format of the background that's the problem.

thyristor59, 2020-09-03:
what would be great is that people can choose the template before taking the photos. 
I don't know how to implement this in the program ....