reuterbal/photobooth #220: Very low resolution rendering

thibv, 2020-06-15:
**Describe the bug**

I'm using the software on a Raspberry Pi 3B+ and a DSLR, Nikon D5300. Control for DSLR is gphoto2 (command line version), which works correctly when used manually with the command line. But when using with the photobooth, the pictures are in very low resolution :
- On the final export, we can clearly see pixels
- When I check photosize/resolution of single shot file, the resolution is 160x120.

**To Reproduce**

1. Connect a DSLR via USB to the Raspberry Pi
2. In config file, choose gphoto2-commandline
3. Use the photobooth
4. Check the rendering quality at the end

**Expected behavior**
Rendering should not be pixelated, and single shot files should be saved as high resolution files (just like when using gphoto2 manually). Note that temporary file stored in /dev/shm/photobooth.jpg seems to be of good quality.

**Hardware (please complete the following information)**

 - Device Raspberry Pi 3B
 - Camera Nikon D5300
 - GPIO: No

**Software (please complete the following information)**

 - OS Raspbian Stretch
 
I'll add output log file and python version this evening, when I'll have the device on hand.

whatzat, 2020-10-04:
Having the same issue. My setup is a Canon 5d classic + Pi4 using the gphoto command line. Camera captured images are 3-4mb while the images capture by the photobooth SW are <1mb. Did you ever find a solution or work around?

thibv, 2020-10-04:
It appears that I've left RAW capture mode on my DSLR, not JPG. So the Pi was using only a preview (low quality) of the RAW file. Changing from RAW to jpg on DSLR resolved the problem.